ros/diagnostics #45: Topics published on a frequency lower than 1.0/diagnostic_period are not supported

peci1, 2016-02-22:
Looking at `TimeStampStatus` and `FrequencyStatus` code (and testing with our hardware), if you use the `TopicDiagnostic` to diagnose a slow topic (0.3 Hz in our case), then the diagnostic fails.

At the time windows a message has arrived, it correctly reports on the frequency and all of this, but at the other time windows when no message has appeared, the diagnostic reports ERROR with `No events recorded. No data since last update.`. 

This is obviously true, but it is not an error state.

peci1, 2016-02-22:
Here's my workaround (tested to work correctly).

https://gist.github.com/peci1/ede392a7006d5a9af6c8

In the workaround, you can also see the basic problem of the provided `TimeStampStatus` and `FrequencyStatus` classes - their members are defined `private` instead of `protected`, effectively disabling people to extend this class and make full use of it.

trainman419, 2016-03-22:
This looks useful, but I don't have a lot of time to integrate it. It will go faster if you submit it as a pull request.

g-gemignani, 2020-08-30:
Closing since the feature has been merged and will be added to both melodic and noetic with the next versions

closed 2020-08-30 by g-gemignani

peci1, 2020-09-20:
I was looking for a good place to improve documentation, but I haven't found any that would specifically cope with FrequencyStatus and TimeStampStatus. Is there such a place that I overlooked?

g-gemignani, 2020-09-22:
Hi @peci1 and thank you for your message. If I am not wrong, in the past the diagnostics maintainers have added documentation to the wiki: http://wiki.ros.org/diagnostics I believe that would be the best place where to add info about your change

peci1, 2022-08-18:
I finally found time to put this into the docs! http://wiki.ros.org/action/info/diagnostic_updater?action=diff&rev2=26&rev1=25

g-gemignani, 2022-08-19:
Thank you for your time!