ros/diagnostics #47: discard_stale parameter ignored in rqt_robot_monitor

pirobot, 2016-05-09:
I'm using ROS Indigo (Debian packages) on Ubuntu 14.04 and I'm trying to use the discard_stale parameter for a number of GenericAnalyzers.    However, even when an analyzer has the discard_stale parameter set to true, the Stale status message appears and persists in the rqt_robot_monitor display.

I have attached a screen shot rqt_robot_monitor showing Stale state of the (non-existent) Base Controller.  And here is my diagnostics.yaml file:

	pub_rate: 1.0 # Optional
	base_path: '' # Optional, prepended to all diagnostic output
	
	analyzers:
	  pub_frequency:
	    type: GenericAnalyzer
	    path: 'Pub Frequency'
	    discard_stale: true
	    timeout: 5.0
	    discard_stale: true
	    contains: 'freq'
	  sensors:
	    type: GenericAnalyzer
	    path: 'Sensors'
	    discard_stale: true
	    timeout: 5.0
	    contains: '_sensor'
	  joints:
	    type: GenericAnalyzer
	    path: 'Joints'
	    discard_stale: true
	    timeout: 1.0
	    regex: '.*_joint$'
	  base_controller:
	    type: GenericAnalyzer
	    path: 'Base Controller'
	    discard_stale: true
	    timeout: 1.0
	    contains: 'base_controller'

[image]

trainman419, 2016-05-09:
The documentation isn't clear on this, but the discard_stale parameter requires at least one diagnostic item within that category to be considered non-stale. This allows individual items to come and go, but to still report stale if the entire category is missing.

The code that implements this is: https://github.com/ros/diagnostics/blob/03e0db006175c3e6157bb50ab38021ebb4995c5b/diagnostic_aggregator/include/diagnostic_aggregator/generic_analyzer_base.h#L150-L194

pirobot, 2016-05-09:
OK, I get it now.  Thanks!

closed 2016-05-09 by pirobot