ros/diagnostics #52: diagnostic_updater needs to check if the diagnostic_period parameter exists

magazino, 2016-09-05:
The current implementation of `diagnostic_updater::Updater` reads the parameter `diagnostic_period` from the parameter server without checking if it exists. If it does not exist, the `getParamCached` method returns 0.0 and update will run every time.

trainman419, 2016-09-05:
I haven't observed this, and I have a lot of code that would probably be susceptible to the bug you're describing.

I had a quick look at the code, and `getParamCached` doesn't modify the value if the parameter doesn't exist, so if the `~diagnostic_period` parameter doesn't exist, `period_` should remain set to the default of 1.0.

Perhaps you can provide a bit of sample code that reproduces this issue?

magazino, 2016-09-05:
We just found out that the root cause for our problem was something else. I'll close this ticket. Sorry for this.

closed 2016-09-05 by magazino