ros/diagnostics #62: Lunar release

mikaelarguedas, 2017-04-10:
Hi @trainman419 and diagnostics maintainers,

As you may know the next ROS release Lunar Loggerhead is around the corner :tada:
Is it possible to release `diagnostics` on ROS Lunar? Being a low level package this is currently [preventing many repositories from being released](http://repositories.ros.org/status_page/blocked_releases_lunar.html?q=id%3D%22diagnostics%22&s=6&r=1).

If you don't have time to make a new release, please release the current kinetic version into Lunar by running `bloom-release diagnostics -r lunar -t lunar --new-track`.
Thanks!

mikaelarguedas, 2017-04-17:
@trainman419 Any update on this? This is currently [the main blocker](http://repositories.ros.org/status_page/blocked_releases_lunar.html?s=6&r=1) for the release of packages into lunar. It would be really awesome to have this repo released into lunar asap. Thanks!

mikaelarguedas, 2017-04-24:
Hi @trainman419 ,
Any update on this ?
If there is no activity on this issue in the next 2days I'll release the current kinetic version into lunar to allow the release to move forward. Please speak up if you want to proceed otherwise.

mikepurvis, 2017-04-24:
Please do tag a new release for Lunar, either 1.9.0 or 1.8.11.

We (Clearpath) have been running the current master branch in our bundles and vouch for its stability.

trainman419, 2017-04-24:
Sorry; I haven't been getting github notifications for this issue for some reason. I'll get moving on a lunar release today.

mikaelarguedas, 2017-04-24:
awesome thanks @trainman419 !

trainman419, 2017-04-25:
I went through and triaged incoming pull requests (github or my email vendor have missed a LOT of notifications), merged the low-risk PRs, made sure all of the tests pass and put out version 1.9.0: https://github.com/ros/rosdistro/pull/14646

mikaelarguedas, 2017-04-25:
:+1: Thank for working on this, all jobs are green :tada:, closing this

closed 2017-04-25 by mikaelarguedas