ros/diagnostics #69: Exception in sensors_monitor / parse_sensor_line

stwirth, 2017-10-09:
I just got this running `sensors_monitor.py` on a ROS kinetic / Ubuntu Xenial machine:
```
[ERROR] [1507561973.045543]: Unable to process lm-sensors data
[ERROR] [1507561973.047346]: Traceback (most recent call last):
  File "/etc/robot/ros/lib/diagnostic_common_diagnostics/sensors_monitor.py", line 189, in monitor
    for sensor in parse_sensors_output(get_sensors()):
  File "/etc/robot/ros/lib/diagnostic_common_diagnostics/sensors_monitor.py", line 156, in parse_sensors_output
    s = parse_sensor_line(line)
  File "/etc/robot/ros/lib/diagnostic_common_diagnostics/sensors_monitor.py", line 109, in parse_sensor_line
    [sensor.name, sensor.type] = name.rsplit(" ",1)
ValueError: need more than 1 value to unpack
```

The output of the `sensors` command is:
```
coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +36.0°C  (high = +84.0°C, crit = +100.0°C)
Core 0:         +33.0°C  (high = +84.0°C, crit = +100.0°C)
Core 1:         +34.0°C  (high = +84.0°C, crit = +100.0°C)
Core 2:         +32.0°C  (high = +84.0°C, crit = +100.0°C)
Core 3:         +30.0°C  (high = +84.0°C, crit = +100.0°C)

nct6106-isa-0290
Adapter: ISA adapter
in0:            +0.72 V  (min =  +0.00 V, max =  +1.74 V)
in1:            +1.66 V  (min =  +0.00 V, max =  +2.04 V)
in2:            +3.41 V  (min =  +0.00 V, max =  +4.08 V)
in3:            +3.33 V  (min =  +0.00 V, max =  +4.08 V)
in4:            +0.63 V  (min =  +0.00 V, max =  +2.04 V)
in5:            +1.66 V  (min =  +0.00 V, max =  +2.04 V)
in6:            +1.70 V  (min =  +0.00 V, max =  +2.04 V)
in7:            +3.07 V  (min =  +0.00 V, max =  +4.08 V)
in8:            +2.03 V  (min =  +0.00 V, max =  +4.08 V)
fan1:          4545 RPM  (min =    0 RPM)
fan2:          2760 RPM  (min =    0 RPM)
fan3:             0 RPM  (min =    0 RPM)
SYSTIN:         +38.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM
                         (crit low = +127.0°C, crit = +127.0°C)  sensor = thermal diode
AUXTIN:         -12.0°C  (high = +80.0°C, hyst = +75.0°C)
                         (crit low = +127.0°C, crit = +127.0°C)  sensor = thermal diode
PECI Agent 0:   +36.0°C  (high = +80.0°C, hyst = +75.0°C)
                         (crit low = +127.0°C, crit = +127.0°C)
PECI Agent 1:    +0.0°C  (high = +80.0°C, hyst = +75.0°C)
                         (crit low = +127.0°C, crit = +127.0°C)
PCH_CHIP_TEMP:   +0.0°C  
PCH_CPU_TEMP:    +0.0°C  
intrusion0:    ALARM
beep_enable:   disabled
```

Looks like the parser cannot read some lines.

VorpalBlade, 2019-10-01:
I got the same. Here is my sensors output:

```
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +45.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +44.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +44.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +45.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +45.0°C  (high = +100.0°C, crit = +100.0°C)

thinkpad-isa-0000
Adapter: ISA adapter
fan1:        2863 RPM

acpitz-virtual-0
Adapter: Virtual device
temp1:        +44.0°C  (crit = +128.0°C)

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +44.0°C  

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +42.5°C  
```

chrisflesher, 2021-02-13:
I'm getting similar issues with:
```
BAT0-acpi-0
Adapter: ACPI interface
in0:          10.00 V  
```