ros/diagnostics #77: Bond broken prevents adding diagnostics

Axel13fr, 2018-02-21:
When adding diagnostics at runtime using either the script manual_diag.py given here http://wiki.ros.org/diagnostics/Tutorials/Adding%20Analyzers%20at%20Runtime or add_analyzers, the aggregator fails to properly update the group and leaves the added diagnostics under /Other/.

Looking at the logs, the aggregator gives:
[ WARN] [1519235533.166823508]: Bond for namespace /startup_analyzers was broken
[ WARN] [1519235533.171578828]: Broken bond tried to remove an analyzer which didn't exist.

I don't see any reasons why as the node adding diags (/startup_analyzers) is still running and I can see /diagnostics_agg/bond pub and sub when rostopic info.

There is only one node adding diags. Any idea or is that directly a Bond issue ?

Axel13fr, 2018-02-22:
Ok I found the issue.

Some changes have been made on the diagnostic aggregator, specifically on the topic used to form the bond which is now "/diagnostics_agg/bond" + "add_diags node namespace".

This change has been made on the aggregator itself but only 2 months later on the add_diags script (so bond topic was not matching) and never in the tutorial.

It didn't make its way to release yet so the current version basically has a mismatch and is not functional.

Can you please proceed to release the fix on add_diags ?

trainman419, 2018-05-02:
Released to Kinetic: https://github.com/ros/rosdistro/pull/17711

closed 2018-05-02 by trainman419