ros/diagnostics #92: Division by zero if time does not change between updates

poggenhans, 2018-11-16:
To determine the update frequency, the diagnostic_updater divides by the time since the last update here: https://github.com/ros/diagnostics/blob/indigo-devel/diagnostic_updater/include/diagnostic_updater/update_functions.h#L174

If the time has not changed since then (happend to me when playing a rosback with `use_sim_time`), we divide by zero.

wkhudgins92, 2019-05-06:
What would the desired behavior be here? I'm not sure that returning any numerical value would be truthful. Perhaps outputting N/A on line 199?

poggenhans, 2019-05-08:
I would simply not add the frequency to the status message if `window` is zero.

wkhudgins92, 2019-05-08:
Valid idea. I'll implement something and raise a PR for this...thanks @poggenhans

closed 2020-08-14 by g-gemignani