ros/diagnostics #106: release a new version

jspricke, 2019-03-25:
Hey Austin,

now with #99 in, can you please do a new release for kinetic and melodic?

Thanks a lot!

Jochen

jspricke, 2019-04-11:
Note that #99 is actually a ABI break due to:

https://github.com/ros/diagnostics/pull/99/files#diff-27d11a6675817723f16ce5fad6c275f8R274

so not sure if we should rather wait for ROS N.

mikepurvis, 2019-04-12:
I'm not an expert on ABI, but this change actually looks like it might be safe, reason being:

- There's no translation unit for `GenericAnalyzerBase`; the whole implementation is in the header, so any other implementations will be self-contained.
- The actual interface is the `Analyzer` class, which is unchanged, so an `Analyzer*` can safely point to instances of `GenericAnalyzerBase` implementations from before or after this change, and won't have any misalignment or even risks of, say calling a destructor with the wrong size.

I say release it into Melodic.

jspricke, 2019-04-12:
I had a similar reasoning when I had a quick look, but debugged a nice segfault yesterday caused by it ;).

mikepurvis, 2019-04-12:
Oh fun. Were you doing anything especially off-beat to cause that?

jspricke, 2019-04-12:
We have an internal package on top of the aggregator using thelibdiagnostic_aggregator.so. Nothing special inside from first look.

mikepurvis, 2019-04-12:
Does it actually have pointers to instances of `GenericAnalyzerBase`, or only to instances of `Analyzer`?

jspricke, 2019-04-12:
`Analyzer` and `GenericAnalyzer`.

mikepurvis, 2019-04-12:
Okay, yeah I can see that being an issue then, since the size of `GenericAnalyzer` would have changed, and in particular, any fields in there would be offset from their previous locations.

Alas.

Alabate, 2019-05-01:
In our case we are also interested in a new release on melodic because of #105 . We're trying to migrate our codebase to python3, and diagnostics are used across the whole project.

flixr, 2019-08-29:
We also need #105 like @Alabate, if the whole thing can't be released as is, could the python3 fixes be backported instead?

g-gemignani, 2020-08-11:
Does someone need also a new release for kinetic? Otherwise, I would close given https://github.com/ros/rosdistro/pull/26156 and https://github.com/ros/rosdistro/pull/26155

closed 2020-08-30 by g-gemignani