ros/diagnostics #127: Python 3 syntax errors, Noetic plan?

sloretz, 2019-10-22:
It looks like there are a few Python 3 syntax errors on the `indigo-devel` branch. Is there a branch for Noetic? I notice there's [no Noetic source entry for the `diagnostics` repo](https://github.com/ros/rosdistro/blob/master/noetic/distribution.yaml), but it looks like it's [recursively blocking 97 other repositories](http://repositories.ros.org/status_page/blocked_source_entries_noetic.html?q=diagnostics&s=6&r=1).

```
$ python3 -m flake8 --select E999
./diagnostic_analysis/scripts/export_csv.py:59:39: E999 SyntaxError: invalid syntax
./diagnostic_analysis/scripts/sparse_csv.py:67:34: E999 SyntaxError: invalid syntax
./diagnostic_analysis/src/diagnostic_analysis/exporter.py:87:51: E999 SyntaxError: invalid syntax
./self_test/scripts/test_selftest.py:81:25: E999 SyntaxError: invalid syntax
./diagnostic_common_diagnostics/src/diagnostic_common_diagnostics/ntp_monitor.py:53:19: E999 SyntaxError: invalid syntax
./diagnostic_common_diagnostics/src/diagnostic_common_diagnostics/hd_monitor.py:365:83: E999 SyntaxError: invalid syntax
./diagnostic_common_diagnostics/src/diagnostic_common_diagnostics/sensors_monitor.py:208:25: E999 SyntaxError: invalid syntax
./diagnostic_aggregator/test/add_analyzers_test.py:136:21: E999 SyntaxError: invalid syntax
./diagnostic_aggregator/test/aggregator_test.py:261:21: E999 SyntaxError: invalid syntax
```

https://github.com/ros/diagnostics/blob/ac407b42e2e27af945d3123517c81974b2ec9986/diagnostic_analysis/scripts/export_csv.py#L59
https://github.com/ros/diagnostics/blob/ac407b42e2e27af945d3123517c81974b2ec9986/diagnostic_analysis/scripts/sparse_csv.py#L67
https://github.com/ros/diagnostics/blob/ac407b42e2e27af945d3123517c81974b2ec9986/diagnostic_analysis/src/diagnostic_analysis/exporter.py#L87
https://github.com/ros/diagnostics/blob/ac407b42e2e27af945d3123517c81974b2ec9986/self_test/scripts/test_selftest.py#L81
https://github.com/ros/diagnostics/blob/ac407b42e2e27af945d3123517c81974b2ec9986/diagnostic_common_diagnostics/src/diagnostic_common_diagnostics/ntp_monitor.py#L53
https://github.com/ros/diagnostics/blob/ac407b42e2e27af945d3123517c81974b2ec9986/diagnostic_common_diagnostics/src/diagnostic_common_diagnostics/hd_monitor.py#L365
https://github.com/ros/diagnostics/blob/ac407b42e2e27af945d3123517c81974b2ec9986/diagnostic_common_diagnostics/src/diagnostic_common_diagnostics/sensors_monitor.py#L208
https://github.com/ros/diagnostics/blob/ac407b42e2e27af945d3123517c81974b2ec9986/diagnostic_aggregator/test/add_analyzers_test.py#L136
https://github.com/ros/diagnostics/blob/ac407b42e2e27af945d3123517c81974b2ec9986/diagnostic_aggregator/test/aggregator_test.py#L261

AutonomousHansen, 2019-11-27:
I'm also concerned about the plan for Python3/Noetic, just posting to show interest

mikepurvis, 2019-12-06:
@sloretz What's the plan from OSRF to move this forward?

sloretz, 2020-03-10:
> @sloretz What's the plan from OSRF to move this forward?

I'm waiting to hear from the maintainers. 

```
diagnostic_aggregator: Brice Rebsamen, Austin Hendrix
diagnostic_analysis: Brice Rebsamen, Austin Hendrix
diagnostic_common_diagnostics: Brice Rebsamen, Austin Hendrix
diagnostics: Brice Rebsamen, Austin Hendrix
diagnostic_updater: Brice Rebsamen, Austin Hendrix
rosdiagnostic: Guillaume Autran
self_test: Brice Rebsamen, Austin Hendrix
test_diagnostic_aggregator: Brice Rebsamen, Austin Hendrix
```

@trainman419 @bricerebsamen @guillaumeautran friendly ping. It looks like `geometry` and `geometry2` are blocking this repo from being released, so it will be this repo's turn soon.

You'll find some useful info about switching to Python 3 on this wiki page http://wiki.ros.org/UsingPython3/

ahcorde, 2020-03-26:
I have been working into bumping the cmake version and some python3 changes https://github.com/ahcorde/diagnostics/tree/ahcorde/noetic_release

Can someone with rights in the repository create a `noetic` branch?

Karsten1987, 2020-03-26:
@ahcorde I've just created a `noetic-devel` branch. Great to hear you picked up some of the works towards noetic.

sloretz, 2020-04-01:
Noetic release in https://github.com/ros/rosdistro/pull/24315

closed 2020-04-01 by sloretz