ros/diagnostics #128: DiscardAnalyzer doesn't work together with GenericAnalyzer

MCFurry, 2019-10-25:
Please see the following example of my `diagnostic_aggregator` configuration:
```
ecu:
  type: diagnostic_aggregator/GenericAnalyzer
  path: 'ECU'
  startswith: 'ecu_reporter'
  remove_prefix: 'ecu_reporter_node: '
ecu_discard1:
  type: diagnostic_aggregator/DiscardAnalyzer
  path: none
  contains: 'Error_676'
ecu_discard2:
  type: diagnostic_aggregator/DiscardAnalyzer
  path: none
  contains: 'Error_1235'

```
The problem seems that no diagnostic messages coming from this node are discarded anymore now that the `GenericAnalyzer` matches them as well and groups them.
There doesn't seem to be any specific ordering in processing `GenericAnalyzer` vs `DiscardAnalyzer`?
Am I miss-using the analyzers or is there a bug here?

In the end I would like to put all diagnostic messages from `ecu_reporter_node` in the same path such that `rqt_robot_monitor` groups them, unless they are discarded of course.

amilcarlucas, 2020-09-21:
This is probably fixed now. Can you update and re-test?

MCFurry, 2020-09-21:
Thx, I'll try to re-evaluate asap!
Is this fix in the apt-released version or do I need to test the latest source?

amilcarlucas, 2020-09-21:
apt-released should have it

yangyy11, 2021-01-31:
> apt-released should have it

> Please see the following example of my `diagnostic_aggregator` configuration:
>
> ```
> ecu:
>   type: diagnostic_aggregator/GenericAnalyzer
>   path: 'ECU'
>   startswith: 'ecu_reporter'
>   remove_prefix: [...]

ok