ros/diagnostics #154: Transfer ownership of ROS1 branches to "orphaned package maintainer group"

Karsten1987, 2020-07-13:
I wanted to start a discussion whether it makes sense to transfer the ownership of the ROS1 branches to the [orphaned package maintainer group](http://wiki.ros.org/OrphanedPackage). 

I personally, together with colleagues at Bosch, will continue to maintain all activity around ROS2, but we'll restrain ourselves from reviewing and releasing patches into ROS1 (i.e. noetic).

Happy to hear opinions from existing maintainers as listed in the package.xml on the metapackage: @trainman419 @bricerebsamen

bricerebsamen, 2020-07-14:
Hello Karsten

I have personally not used ROS in years, and certainly not maintained that
package in a very long time. @trainman419 took care of most of the build
issues in the last few years. AFAIC you can remove me from the list of
maintainers and do whatever you want with this package.

Brice

g-gemignani, 2020-07-18:
Hello, as I mentioned in another issue, at Magazino we are using ros1 diagnostic. I imagine us maintaining it if no one else can take over this chore

Karsten1987, 2020-07-20:
Absolutely appreciated @g-gemignani. I'd propose you open a PR updating the current ROS1 branch (in this case `noetic` I would say) and mark yourself or your team as the maintainer of the packages listed in there. 
As stated above, I guess we could remove Brice as a maintainer in the same run.

bricerebsamen, 2020-07-20:
yes

closed 2020-08-11 by g-gemignani