ros/diagnostics #156: [ROS2 | Foxy] diagnostic_aggregator example crashes because example_pub.py is not installed as a program

airballking, 2020-08-01:
It seems the example publisher of the diagnostic_aggregator publisher is not installed as a program. Hence, the overall example fails.

I ran this on ROS2 Foxy installed from Ubuntu version: 20.04 deb files. Last update was August 1st, 2020.

Here is my error description:

```shell
$ ros2 launch diagnostic_aggregator example.launch.py
[INFO] [launch]: All log files can be found below /home/myuser/.ros/log/2020-08-01-13-22-57-725483-mypc-30482
[INFO] [launch]: Default logging verbosity is set to INFO
/opt/ros/foxy/share/diagnostic_aggregator/example.launch.py:9: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  aggregator = launch_ros.actions.Node(
/opt/ros/foxy/share/diagnostic_aggregator/example.launch.py:14: UserWarning: The parameter 'node_executable' is deprecated, use 'executable' instead
  diag_publisher = launch_ros.actions.Node(
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'example_pub.py' not found on the libexec directory '/opt/ros/foxy/lib/diagnostic_aggregator' 
Task exception was never retrieved
future: <Task finished name='Task-8' coro=<LaunchService._process_one_event() done, defined at /opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py:271> exception=RuntimeError('Signal event received before subprocess transport available.')>
Traceback (most recent call last):
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 273, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/launch_service.py", line 293, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/opaque_function.py", line 75, in execute
    return self.__function(context, *self.__args, **self.__kwargs)
  File "/opt/ros/foxy/lib/python3.8/site-packages/launch/actions/execute_process.py", line 421, in __on_signal_process_event
    raise RuntimeError('Signal event received before subprocess transport available.')
RuntimeError: Signal event received before subprocess transport available.
[INFO] [aggregator_node-1]: process started with pid [30484]
[aggregator_node-1] [INFO] [1596288177.788118179] [AnalyzerGroup]: Retrieved 17 parameter(s) for analyzer group with prefix ''.
[aggregator_node-1] [INFO] [1596288177.788164907] [AnalyzerGroup]: Group '//Example', creating diagnostic_aggregator/GenericAnalyzer 'FindAndRemove' (breadcrumb: contains) ...
[aggregator_node-1] [INFO] [1596288177.788451123] [GenericAnalyzerBase]: Initialized analyzer 'FindAndRemove' with path '//Example/FindAndRemove' and breadcrumb 'contains'.
[aggregator_node-1] [INFO] [1596288177.788457465] [AnalyzerGroup]: Adding analyzer 'FindAndRemove' to group '//Example'.
[aggregator_node-1] [INFO] [1596288177.788461974] [AnalyzerGroup]: Group '//Example', creating diagnostic_aggregator/DiscardAnalyzer 'DoesItNeedAPart?' (breadcrumb: discard) ...
[aggregator_node-1] [INFO] [1596288177.788515414] [GenericAnalyzerBase]: Initialized analyzer 'DoesItNeedAPart?' with path '//Example/DoesItNeedAPart?' and breadcrumb 'discard'.
[aggregator_node-1] [INFO] [1596288177.788518319] [AnalyzerGroup]: Adding analyzer 'DoesItNeedAPart?' to group '//Example'.
[aggregator_node-1] [INFO] [1596288177.788521776] [AnalyzerGroup]: Group '//Example', creating diagnostic_aggregator/GenericAnalyzer 'FindAndRemove' (breadcrumb: findandremove) ...
[aggregator_node-1] [INFO] [1596288177.788532676] [GenericAnalyzerBase]: Initialized analyzer 'FindAndRemove' with path '//Example/FindAndRemove' and breadcrumb 'findandremove'.
[aggregator_node-1] [INFO] [1596288177.788535331] [AnalyzerGroup]: Adding analyzer 'FindAndRemove' to group '//Example'.
[aggregator_node-1] [INFO] [1596288177.788538387] [AnalyzerGroup]: Group '//Example', creating diagnostic_aggregator/GenericAnalyzer 'Level0' (breadcrumb: level0) ...
[aggregator_node-1] [INFO] [1596288177.788547113] [GenericAnalyzerBase]: Initialized analyzer 'Level0' with path '//Example/Level0' and breadcrumb 'level0'.
[aggregator_node-1] [INFO] [1596288177.788549678] [AnalyzerGroup]: Adding analyzer 'Level0' to group '//Example'.
[aggregator_node-1] [INFO] [1596288177.788553035] [AnalyzerGroup]: Group '/Example', creating diagnostic_aggregator/GenericAnalyzer 'Regex' (breadcrumb: regex) ...
[aggregator_node-1] [INFO] [1596288177.788597899] [GenericAnalyzerBase]: Initialized analyzer 'Regex' with path '//Example/Regex' and breadcrumb 'regex'.
[aggregator_node-1] [INFO] [1596288177.788600564] [AnalyzerGroup]: Adding analyzer 'Regex' to group '/Example'.
[aggregator_node-1] [INFO] [1596288177.788603219] [AnalyzerGroup]: Initialized analyzer group '/Example' with path '//Example' and breadcrumb ''.
[aggregator_node-1] [INFO] [1596288177.788608038] [GenericAnalyzerBase]: Initialized analyzer 'Other' with path '//Example/Other' and breadcrumb ''.
[ERROR] [aggregator_node-1]: process[aggregator_node-1] failed to terminate '5' seconds after receiving 'SIGINT', escalating to 'SIGTERM'
[INFO] [aggregator_node-1]: sending signal 'SIGTERM' to process[aggregator_node-1]
[ERROR] [aggregator_node-1]: process has died [pid 30484, exit code -15, cmd '/opt/ros/foxy/lib/diagnostic_aggregator/aggregator_node --ros-args --params-file /opt/ros/foxy/share/diagnostic_aggregator/example.yaml'].
```

It seems the dummy publisher is not found. When I try to run, it fails:
```shell
$ ros2 run diagnostic_aggregator example_pub.py
No executable found
```

However, the script seems to be in the correct directory:
```shell
$ ls /opt/ros/foxy/lib/diagnostic_aggregator
aggregator_node  example_pub.py
```

So, something else must be up. It seems, the program is not flagged as executable and hence not found:
```shell
$ ls -ll /opt/ros/foxy/lib/diagnostic_aggregator/
total 28
-rwxr-xr-x 1 root root 18704 Jul 22 05:20 aggregator_node
-rw-r--r-- 1 root root  4560 Jul 10 02:38 example_pub.py
```

I suspect that is due to [this line in the CMakeLists](https://github.com/ros/diagnostics/blob/foxy/diagnostic_aggregator/CMakeLists.txt#L127):

```
install( # example publisher
  FILES example/example_pub.py
  DESTINATION lib/${PROJECT_NAME}
)
```

If in an overlay, I change this line and re-compile, it works:
```
install( # example publisher
  PROGRAMS example/example_pub.py
  DESTINATION lib/${PROJECT_NAME}
)
```

@Karsten1987 Could you please confirm this issue and that my solution is correct? If both yes, I am happy to provide a short pull request. ;)

airballking, 2020-08-01:
Silly me. It such a small PR, I just did it right away. No need to waste more time.

airballking, 2020-08-05:
I guess this issue stays open until the patch has been back-ported from the ros2-devel branch.

Karsten1987, 2020-08-05:
your PR was already backported and released: https://github.com/ros/diagnostics/pull/160
I am closing this ticket, please don't hesitate to report if your problem isn't addressed yet.

closed 2020-08-05 by Karsten1987

airballking, 2020-08-05:
Cool.