ros/diagnostics #165: Ros2 pkg list does not have diagnostic_updater

Melvin-leo, 2020-10-02:
Hi everyone, I am relatively new to Ubuntu. Currently, I am trying to install diagnostic_updater package with colcon build for eloquent ROS 2. However, only diagnostic_msgs is there in pkg list and not diagnostic_updater. May I know how can I build in the way that my other application can acces diagnostic_updater?

norro, 2020-10-02:
Hi @Melvin-leo. The diagnostic_updater package is part of the ros distro, see [eloquent/distribution.yaml#L521](https://github.com/ros/rosdistro/blob/master/eloquent/distribution.yaml#L521) and is also available as ubuntu (debian) package, see [index.ros.org/diagnostic_updater#eloquent](https://index.ros.org/p/diagnostic_updater/github-ros-diagnostics/#eloquent).

The `diagnostic_msgs` package you are referring to is part of the [common_interfaces](https://github.com/ros2/common_interfaces) repository, which comes with the default ROS 2 repos. So since you are seeing that already but not the updater, I suspect you did not clone [this diagnostic repository](https://github.com/ros/diagnostics) into your workspace, yet. Once you cloned it you can run: `colcon build --packages-up-to diagnostic_updater` in your ROS 2 workspace and should get the package installed.

For general questions regarding packaging and installation, I suggest to visit https://answers.ros.org/questions/

Melvin-leo, 2020-10-02:
Do i need to switch the branch to eloquent?

Karsten1987, 2021-03-03:
@Melvin-leo I assume your question was answered. I am going to close this issue. Please feel free to re-open this ticket if you feel there's more to do here which can't be addressed through ros-answers.

closed 2021-03-03 by Karsten1987