ros/diagnostics #166: bondcpp dependency issue

Chris-Bee, 2020-10-31:
Hello everybody,

I'm using diagnostic_aggregator, branch 'indigo-devel' as an automatically referenced dependency of mavlink and mavros with ROS melodic. While compiling I get the issue bondcppConfig.cmake could not be found. However, ros-melodic-bondcpp is installed.

I did not get this error about 2 months ago and I'm not sure if this is an issue related to bondcpp or the diagnostic_aggregator.

Thanks in advance.

g-gemignani, 2020-11-02:
Hi Chris and thank you for your message. 

I must admit it is a bit hard to debug your issue without having a minimal use case at hand.
In any case, the first thing I would try is to clone different indigo-devel branches and see if some of them fix your issue: I would try with the current master (I didn't manage to release it yet but there were a couple of fixes pushed to it) and an earlier version of it (e.g., 1.9.5). After having tried that, with the experiment results we can try to investigate more in depth.
Have a great day! :)

Chris-Bee, 2021-03-05:
@g-gemignani I agree, apologies for the sparse information. In the end, we think that this was some kind of odd behavior by the setup on our side. We were using a dockerfile to set up a test environment and between installing and compiling mavlink and ros-melodic-bondcpp, there must have been a missing resourcing of the ros setup. With this, I assume the issue can be closed.

g-gemignani, 2021-03-05:
@Chris-Bee no worries. Closing the issue then :)

closed 2021-03-05 by g-gemignani