ros/diagnostics #167: ROS2 : queue size for subscription in DiagnosticAggregator

mitsudome-r, 2020-11-06:
### Current Implementation
The queue size for `/diagnostics` topic is set to SystemDefault in ROS2, whereas it was explicitly set to 1000 in ROS1. 

ROS2: https://github.com/ros/diagnostics/blob/48e16167bdef2760ce0536f226cad264a607b96d/diagnostic_aggregator/src/aggregator.cpp#L99-L100
ROS1: https://github.com/ros/diagnostics/blob/9a3748a224a1a28b7357a3cd9958fc4a9c599354/diagnostic_aggregator/src/aggregator.cpp#L66

### Potential Issue
This may cause problem if system default's history_depth(queue size) is too small and if there are multiple nodes publishing the `/diagnostics` topic.
For example, if system default is set to 1 and if there are multiple sensor driver nodes running on a system each publishing `diagnostics` topic, we can only process DiagnosticStatus from one of the sensors at [`spin_some()`](https://github.com/ros/diagnostics/blob/master/diagnostic_aggregator/src/aggregator_node.cpp) function, which could overlook errors in other sensors.

### Proposal
I think it would be more convenient if we create a new parameter to specify history_depth(queue size) for the subscription. Other wise we have to modify SystemDefault configuration for DiagnosticAggregator, which could potentially affect performance of other nodes that use system_default settings.
e.g.
```
diag_sub_ = n_->create_subscription<DiagnosticArray>(
    "/diagnostics", rclcpp::SystemDefaultsQoS().keep_last(history_depth), std::bind(&Aggregator::diagCallback, this, _1));
```

If we can agree to go with this, I'm willing to create PR to do this modification.
Any comments/suggestions?

mitsudome-r, 2020-11-11:
@Karsten1987 @norro @1r0b1n0
It looks like you three did the porting to ROS2.
Do you have any comments about this?

1r0b1n0, 2020-11-12:
Hi,
I completely agree this is a problem because the default QoS history size is 1 (at least it was in eloquent).
In one of my projects I changed this line to
```
diag_sub_ = n_->create_subscription<diagnostic_msgs::msg::DiagnosticArray>("/diagnostics",
      rclcpp::QoS(1000), std::bind(&Aggregator::diagCallback, this, _1));
```
I think it would be better to, at least by default, set the history to a large number (like 1000 -> this means we support up to 1000 nodes publishing a /diagnostics) so we don't have to tune the DiagnosticsAggregator for each project.

norro, 2020-11-16:
@mitsudome-r Agreed!

cdbierl, 2021-01-18:
Even with the recent history depth adjustment to 1000, the diagnostic aggregator still appears to fall behind with my application. I am basically comparing the rqt_runtime_monitor to the rqt_robot_monitor's display. The runtime monitor reveals the latest diagnostic entry but it is not being mirrored in the robot monitor pane. A simple diagnostic updater exposing a integer counter will reveal the drift.

For my application, when I run ```ros2 topic hz /diagnostics```, it shows around a 5 Hz message rate. Perhaps the diagnostic aggregator is being overworked?

cdbierl, 2021-01-19:
I found a remedy for the behavior I described above. It's not that aggregator is being "overworked," its subscription callback is not being given enough time to run, so the published output never reflects the current state of the diagnostics. See my PR:  https://github.com/ros/diagnostics/pull/179

Karsten1987, 2021-03-03:
I am closing this as the issue has been addressed in #168

closed 2021-03-03 by Karsten1987