ros/diagnostics #172: On Foxy throws exception using diagnostic_aggregator tutorial

anderwm, 2021-01-04:
Not sure if the tutorial(https://wiki.ros.org/diagnostics/Tutorials/Configuring%20Diagnostic%20Aggregators) just needs updating or if the code could do some smart casting, but...

Specifically the 'contains' field must be a list of strings now now contains: ['whatever'] instead of contains: 'whatever. Also crashes if timeout is an int instead of double.

norro, 2021-01-11:
Thanks for reporting this issue!

The tutorial indeed needs an update for ROS 2. @Karsten1987 do you have an idea how to go forward with this or any common practice how you handled this with other packages? Should be split the tutorial in a ROS 1 and ROS 2 version? Or should we mark the existing tutorial as _'ROS 1 only'_ and add a ROS 2 tutorial at a different place later?

@anderwm Until this is sorted out, you may want to have a look at the foxy [diagnostic aggregator example](https://github.com/ros/diagnostics/tree/foxy/diagnostic_aggregator/example).

Karsten1987, 2021-01-11:
I generally prefer to have the documentation close to the code. That is, have the tutorial be placed in a README.rst or similar in the same repo. That allows to adapt quickly to differences between branches etc. 
In this particular case, I'd say we write an appropriate tutorial for the ROS2 version of diagnostics and pack it in a readme. Once it's released, this readme should show up on index.ros.org to which we could link a note from ros.wiki.org indicating that the documentation for ROS2 can be found there.

Opinions?

ct2034, 2023-06-12:
Closing this since foxy is EOL. If this occurs in a newer release, please reopen.

closed 2023-06-12 by ct2034