ros/diagnostics #185: ROS1 hd_monitor.py fails on noetic

chrisflesher, 2021-02-12:
Am getting the following traceback:

```
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/diagnostic_common_diagnostics/sensors_monitor.py", line 46, in <module>
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/diagnostic_common_diagnostics/hd_monitor.py", line 361, in <module>
    from StringIO import StringIO
ModuleNotFoundError: No module named 'StringIO'
    hostname_clean = string.translate(hostname, string.maketrans('-','_'))
AttributeError: module 'string' has no attribute 'translate'
```

chrisflesher, 2021-02-12:
Also am getting the following error after running sensors_monitor.py:

```
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/diagnostic_common_diagnostics/sensors_monitor.py", line 46, in <module>
    from StringIO import StringIO
ModuleNotFoundError: No module named 'StringIO'
```

g-gemignani, 2021-02-13:
Hi @chrisflesher and thank you for your message.
It looks like those nodes were never completely ported to python 3. I quickly did it. Can you please double-check that they are now working? You can find the fixes in https://github.com/ros/diagnostics/tree/noetic/fix-diagnostic-common-diagnostics
If it works for you I'll take care of creating a new release :)

chrisflesher, 2021-02-13:
Thanks! I was able to get `hd_monitor.py` working after making the following fix:
```
diff --git a/diagnostic_common_diagnostics/src/diagnostic_common_diagnostics/sensors_monitor.py b/diagnostic_common_diagnostics/src/diagnostic_common_diagnostics/sensors_monitor.py
index e26fc8d..ffebb65 100755
--- a/diagnostic_common_diagnostics/src/diagnostic_common_diagnostics/sensors_monitor.py
+++ b/diagnostic_common_diagnostics/src/diagnostic_common_diagnostics/sensors_monitor.py
@@ -150,7 +150,7 @@ def _rpm_to_rads(rpm):
     return rpm * (2 * math.pi) / 60
 
 def parse_sensors_output(output):
-    out = StringIO(output)
+    out = StringIO(output.decode('utf-8'))
 
     sensorList = []
     for line in out.readlines():
```

I tried fixing `sensors_monitor.py` but found that code may need a significant rewrite to be robust. I'm running an Ubuntu 20.04 VM and its crashing for multiple reasons with the following `sensors` output:
```
BAT0-acpi-0
Adapter: ACPI interface
in0:          10.00 V  
```

chrisflesher, 2021-02-13:
I think I'd be pretty happy with the fix for `hd_monitor.py` now, maybe we could try to fix `sensors_monitor.py` as a separate issue? Setting up some unit tests might be helpful, I could take a crack at that if I get some time to work on it.

chrisflesher, 2021-02-13:
The sensors issue seems like a duplicate of #69

g-gemignani, 2021-02-14:
Hi @chrisflesher, you are right, there was still an issue with sensors_monitor.py. Can you try now?

chrisflesher, 2021-02-18:
Yes, it seems to run ok now. Thanks!

chrisflesher, 2021-02-18:
A new release of this would be great. I have only tested it on a VM, could test on actual hardware next week and report results if that would be helpful.

g-gemignani, 2021-02-20:
We are in no rush. Let's wait for your additional test and then I take care of merging and releasing. Thank you!

g-gemignani, 2021-03-05:
@chrisflesher any update on this?

g-gemignani, 2021-03-14:
Merged to noetic-devel. Closing the issue

closed 2021-03-14 by g-gemignani