ros/diagnostics #188: [ROS2] diagnostic_updater::Updater has not method for caller to control update()

tgreier, 2021-02-24:
I would like to have control over the threading of the diagnostics update, instead of using a node timer (which could block other callbacks).
For diagnostics updater, I'd want option to create a low-level thread that will handle the call to update().

I can do a pull request to implement this, but I would like to know if there was a reason for this change from ROS1->ROS2.
ROS1 gave the responsibility of calling update() to the application instead of creating a timer internal to the updater.

Karsten1987, 2021-02-26:
I guess that's a fair use-case. The timer seemed to me the most ROSifed way to handle a constant update rate, but I totally see your need for taking care of threading yourself.

If you have already an idea in mind, please go ahead and draft a PR. I will say though that the timer approach should still be supported, in a way to have a "fire and forget" solution for it, where users don't have to deal with threading themselves.

As a side note, there is a new callback group executor solution in ROS right now, which lets you add callbacks to different callback groups so that these groups don't block each other. Maybe that's a solution for you as well? For more details about it see https://github.com/ros2/rclcpp/pull/1218

Timple, 2025-11-04:
We could specify period=0 to disable timer creation.

Than folks can call `force_update()` as they see fit. Would only take a small modification I guess.