ros/diagnostics #195: missing messages with multiple diagnostic updaters

doronhi, 2021-04-19:
I want to publish diagnostics on several objects.
To that end, I create several objects of type `diagnostic_updater::Updater`, each of them is being ticked() and updated().
When I listen on the `/diagnostics` topic I notice that sometimes one of the objects fails to update for a cycle or several cycles.
I expect the `/diagnostics` topic to contain a message for every updater every second.
I consider a "jump" to be an event where the updater fails to publish the diagnostics message for more than 1.5 seconds (the message is published once a second so 1.5 is a threshold to catch a gap of 2 seconds and more)

To test, I wrote a tester, based on the ["talker" demo](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29) in the attached file: [talker.cpp](https://github.com/doronhi/python_utils/blob/master/talker.cpp)
To parse the output of `rostopic echo /diagnostics` I wrote a python application that prints the number of "jumps" for every updater. The parsing app is [parse_rostopic_echo.py](https://github.com/doronhi/python_utils/blob/master/parse_rostopic_echo.py)

To test run the followings:
In terminal 1:
```
roscore
```

In terminal 2:
```
rostopic echo /diagnostics > diagnostics_out.txt
```

In terminal 3:
```
rosrun beginner_tutorials talker 5
```
(The last parameter - 5 is the number of updaters)
talker node will run for 30 seconds and stop. When it does, stop the `rostopic echo` command in terminal 2 (CTRL-C) and run the following:
```
python parse_rostopic_echo.py diagnostics_out.txt --show_jumps "talker_test_diag_[^:]*"
```

The output I get when running with 5 updaters is something like this:
```
talker_test_diag_0:0:[]
talker_test_diag_1:7:[ 6.39909482  2.00166702  2.10049343  4.19825268  2.09969783  3.20129704  3.19923377]
talker_test_diag_2:6:[ 8.40078616  2.10051894  2.00056171  2.09968781  4.30027509  2.1004262 ]
talker_test_diag_3:1:[ 4.19952869]
talker_test_diag_4:0:[]
```
It means, for every updater, its name: number of jumps: time difference between consequtive updates.

The output when running with 1 updater is:
```
talker_test_diag_0:0:[]
```

Is the usage of updater in talker.cpp wrong?
Is this behavior known or can I expect that for 5 updaters or more I will still get a message every second for every updater.

Thanks

g-gemignani, 2021-04-21:
Hi @doronhi and thank you for your message. 
Unfortunately, all the links you provided give me a 404 error. Can you please fix it so that I can try to support you?
Thank you! :)

doronhi, 2021-04-21:
Sorry. 
Please extract the attached **beginner_tutorials.zip** file to your `catkin_ws\src` folder. 
**parse_rostopic_echo.zip** contains just parse_rostopic_echo.py so you can extract it anywhere you want.

[parse_rostopic_echo.zip](https://github.com/ros/diagnostics/files/6348968/parse_rostopic_echo.zip)
[beginner_tutorials.zip](https://github.com/ros/diagnostics/files/6348969/beginner_tutorials.zip)

g-gemignani, 2021-05-02:
Hi @doronhi and sorry if it took me a bit to find the time to answer you.
I played a bit with your code and the issue that you described. 
I noticed that adding a small `ros::Duration(0.001).sleep();` before calling the update function (line 68 of your talker file),  makes the issue disappear.
I might be wrong, but I remember that in ROS, sometimes msgs are dropped when someone tries to publish msgs in a row too fast. I suspect that this issue has to do something with the ROS communication infrastructure, rather than the diagnostics library itself.
I hope that this message helps you out and clarifies a bit the behavior of your code!

doronhi, 2021-05-04:
Thanks for looking into it. I'll test the workaround and report the results.

guillaumeautran, 2021-05-04:
Isn't the problem that the publisher to the `/diagnostics` topic is set to a queue of 1? 
https://github.com/ros/diagnostics/blob/noetic-devel/diagnostic_updater/include/diagnostic_updater/diagnostic_updater.h#L556

If we increase this queue depth, then the messages will not be dropped.  Can this be done?

g-gemignani, 2021-05-04:
Hi @guillaumeautran,
thank you for pointing that out. I didn't think of the queue of the updater, but it definitely makes sense. Let me try that out and see if this fixes the problem. I'll come back to you as soon as I find a bit of time.

g-gemignani, 2021-05-04:
Hello, I managed to find 5 min of time before I thought I would.
Unfortunately, even extending the queue of the publisher to 100, I personally still get the same issue:

> $ python parse_rostopic_echo.py diagnostics_out.txt --show_jumps "talker_test_diag_[^:]*"
> talker_test_diag_0:0:[]
> talker_test_diag_1:1:[2.19938827]
> talker_test_diag_2:7:[2.19949651 4.3000288  4.30032825 2.10019279 2.19931293 2.10009646 2.10079837]
> talker_test_diag_3:5:[2.10022593 2.19935894 2.19940019 3.19998956 2.10089087]
> talker_test_diag_4:0:[]

If I find time, I can try to follow this train of thoughts and see if it leads to something...

guillaumeautran, 2021-05-06:
I'm not sure the test app accurately represents the real world scenario. 
Looks like the loop rate is 10Hz. But diagnostics are usually published at a 1Hz frequency. 
So, what happens if the publication rate is dropped down to 1Hz and the publish count increased to 10 (as an example). With a topic queue of 10, we should not be loosing any message.

doronhi, 2021-05-19:
The test was built specifically to challenge the system and to demonstrate a phenomenon we noticed when publishing diagnostics for several topics at 1Hz each.
The fix/workaround suggested by @g-gemignani, adding a 1-millisecond sleep between consecutive updates may not be ideal but it works for me.
Thanks.

closed 2021-05-19 by doronhi