ros/diagnostics #196: Generic analyzer outputs OK status with ERROR status input

mortenfyhn, 2021-04-26:
If I output `/diagnostics` with a DiagnosticArray with an ERROR status followed by an OK status, the generic analyzer's aggregated output is the OK status. It should be the more severe ERROR status. However, if I flip the order, with the OK first, then the ERROR, I get the correct output. So it appears that the generic analyzer is sensitive to the order of the incoming diagnostic statuses. This is probably not the expected behaviour.

---

Here's an example:

If I configure an aggregator like so

```yaml
analyzers:
  foo:
    type: diagnostic_aggregator/GenericAnalyzer
    path: Foo
    name: "foo"
```

and publish `/diagnostics` like so

```yaml
header: [...]
status:
  -
    level: 2
    name: "foo"
    message: "Error"
    hardware_id: "none"
    values: []
  -
    level: 0
    name: "foo"
    message: "Ok"
    hardware_id: "none"
    values: []
```

then the output on `/diagnostics_agg` misses the error level diagnostic:

```yaml
header: [...]
status:
  -
    level: 0
    name: "/Foo"
    message: "OK"
    hardware_id: ''
    values:
      -
        key: "foo"
        value: "Ok"
  -
    level: 0
    name: "/Foo/foo"
    message: "Ok"
    hardware_id: "none"
    values: []
```

And again, if I flip the order, it works, and I get the error on `/diagnostics_agg`.

g-gemignani, 2021-05-01:
Hi @mortenfyhn and thank you for your message,

the generic analyzer is thought to be used with different messages published onto /diagnostics and not with different statuses added to the same message. By publishing two different and distinct messages on /diagnostics (ideally with two different names and a rule for matching them added to the yaml) you will get the wanted behavior. I hope this helps a bit! :)

mortenfyhn, 2021-05-02:
Thank you for the answer.

I'm a little confused. Since /diagnostics takes DiagnosticArray messages, and DiagnosticArray takes an array of DiagnosticStatuses, it seems odd to require only one status per message. If I have, say, a hardware device driver, and it can have various states such as "temperature too high" or "communication error", is it wrong to create one DiagnosticStatus for each of those two, and publish them as one DiagnosticArray?

g-gemignani, 2021-05-02:
It is not wrong, but it is wrong to publish them all with the same name. Ideally, you would have to publish the first one as foo/something and the second one as foo/something_else and then use the regex matching to aggregate them both under foo.
Please let me know if I explained myself clear enough, otherwise let me know that I try to do better :)

mortenfyhn, 2021-05-03:
Aha! So something like this might be more correct?

```yaml
header: [...]
status:
  -
    level: 2
    name: "my_sensor/temperature"
    message: "Too hot"
    hardware_id: ""
    values: []
  -
    level: 0
    name: "my_sensor/communication"
    message: "Connected"
    hardware_id: ""
    values: []
```

Would you want a patch where I make the generic analyzer output a little warning to the console if it receives multiple statuses with the same name? That might make it easier to people to avoid the same mistake.

g-gemignani, 2021-05-03:
Correct!
A patch would be lovely! 
Thanks!
Gu

mortenfyhn, 2021-05-03:
Great, thank you!

closed 2021-05-03 by mortenfyhn