ros/diagnostics #199: error: ‘no_data_is_problem’ does not name a type

sibi-venti, 2021-06-18:
**I use Ros kinetic, i am getting this problem when i compile it , i tried inidigo devel, and noetic devel both have same problems**
```
 In member function ‘virtual void diagnostic_updater::TimeStampStatus::run(diagnostic_updater::DiagnosticStatusWrapper&)’:
/home/username/catkin_venti/src/diagnostics/diagnostic_updater/src/timestamp_status.cpp:13:18: error: ‘no_data_is_problem’ does not name a type
       const auto no_data_is_problem = dynamic_cast<SlowTimeStampStatus*>(this) == nullptr;
                  ^
/home/username/catkin_venti/src/diagnostics/diagnostic_updater/src/timestamp_status.cpp:14:18: error: ‘status’ does not name a type
       const auto status = no_data_is_problem ? DiagnosticStatus::WARN : DiagnosticStatus::OK;
                  ^
/home/username/catkin_venti/src/diagnostics/diagnostic_updater/src/timestamp_status.cpp:15:20: error: ‘status’ was not declared in this scope
       stat.summary(status, "No data since last update.");
                    ^
diagnostics/diagnostic_updater/CMakeFiles/diagnostic_updater.dir/build.make:62: recipe for target 'diagnostics/diagnostic_updater/CMakeFiles/diagnostic_updater.dir/src/timestamp_status.cpp.o' failed
make[2]: *** [diagnostics/diagnostic_updater/CMakeFiles/diagnostic_updater.dir/src/timestamp_status.cpp.o] Error 1
CMakeFiles/Makefile2:2078: recipe for target 'diagnostics/diagnostic_updater/CMakeFiles/diagnostic_updater.dir/all' failed
make[1]: *** [diagnostics/diagnostic_updater/CMakeFiles/diagnostic_updater.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 29%] Linking CXX executable /home/username/catkin_venti/devel/lib/diagnostic_updater/example
[ 29%] Built target example
[ 32%] Linking CXX shared library /home/username/catkin_venti/devel/lib/libdiagnostic_aggregator.so
[ 32%] Built target diagnostic_aggregator
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed

```

g-gemignani, 2021-06-21:
Hi @sibi-venti and thank you for your message. Would it be possible to get a bit more info on the computer setup you are using? Could it be an issue with the compiler you are using or with your environment setup?

sibi-venti, 2021-06-21:
@g-gemignani you can close the issue here, for kinetic , Groovy-devel works fine for me now

closed 2021-06-21 by sibi-venti