ros/diagnostics #232: [foxy - python] Is it normal that a STEADY_TIME clock is used in TimeStampStatus ?

LoicFerrot, 2022-07-05:
Hello!

I'm wondering why the clock source is set on `STEADY_TIME` and not on `ROS_TIME` at [this line](https://github.com/ros/diagnostics/blob/0145a835db41addaef24aecdc4c776005b9b6a1d/diagnostic_updater/diagnostic_updater/_update_functions.py#L197) in `TimeStampStatus` in the foxy branch.

From my understanding, reading the [clock and time ROS2 article](https://design.ros2.org/articles/clock_and_time.html) that redirects to the [steady_time C++ reference](https://en.cppreference.com/w/cpp/chrono/steady_clock), the steady clock is "most suitable for measuring intervals", so it is perfect for the `FrequencyStatus` class for example. However since a message and more precisely its timestamp can be created by any device in the ROS network, comparing two steady clock times on different devices does not make sense in my opinion since "This clock is not related to wall clock time (for example, it can be time since last reboot)"

Am I missing something?

Thanks for your help!

ct2034, 2022-11-21:
This will be fixed by #214

ct2034, 2022-11-21:
Or do you need it specifically for foxy?

LoicFerrot, 2022-11-23:
Yes, if this is not too much work.

ct2034, 2022-11-23:
Okay, sorry. The mentioned pr replaced STEADY_TIME at another place. This is still open. And I will also backport it.

And just to be verbose: I don't know why STEADY_TIME was used here. But after reading https://design.ros2.org/articles/clock_and_time.html I think it should be ROS_TIME instead. The rationale would be that diagnostic events should be timed relative to the rest of the robot such that if you were to record them in a bag file the can be analyzed in comparison to other data that uses the ROS time.
There may be points to be made for the other two types of time but I think this is the most common use case.

ct2034, 2022-12-16:
Fixed in https://github.com/ros/diagnostics/pull/214

closed 2023-02-13 by ct2034