ros/diagnostics #235: How to Creating a Diagnostic Analyzer with ROS 2 (galactic)?

Li-Jun-SD, 2022-08-25:
Hello,
http://wiki.ros.org/diagnostics/Tutorials/Creating%20a%20Diagnostic%20Analyzer is not work with ROS 2.
Could you write a tutorial about how to write an analyzer plugin in ROS 2? The virtual function interface's change and colcon build system make me feel confuse.
Very appreciate for your help.
Sorry for my English, hope you could understand my words.

Thanks for your work.

Li-Jun-SD, 2022-09-02:
I made a demo based on that tutorial could work with ros2.
[](https://github.com/Li-Jun-SD/pr2_motors_analyzer_ros2)
I wish that could help who have the same problem.

ralph-lange, 2022-11-21:
Thank you for publishing this demo. In my opinion, this issue can be closed now but you may consider to contribute a tutorial for [docs.ros.org](https://docs.ros.org/), by a PR to https://github.com/ros2/ros2_documentation.

closed 2022-12-05 by Li-Jun-SD