ros/diagnostics #262: Humble diagnostic_common_diagnostics build error

xkaraman, 2022-12-12:
For some reason `package.xml` of this package has `catkin` `buildtool` tag that seems off for a ROS 2 package. It produces error such as 

```
The manifest of package "diagnostic_common_diagnostics" (with format version 1) must not contain the following tags: exec_depend.
Either update to a newer format or replace <exec_depend> tags with <run_depend> tags.
```

I think the correct way is just like `foxy` and `galactic` branch meaning the `buildtool` tag must be `ament_cmake` and `ament_cmake_python`. A pull request have been opened by another user as well that fixes it.

ct2034, 2022-12-12:
Thanks for reporting @xkaraman ! These must be a remainder of an uncomplete port.

sumedhreddy90, 2022-12-13:
@ct2034 Even I am facing the same issue

```
--- stderr: diagnostic_common_diagnostics                                                                          
Error parsing '/home/sumedh/mav_platform_r2/src/diagnostics/diagnostic_common_diagnostics/package.xml':
Traceback (most recent call last):
  File "/opt/ros/humble/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 150, in <module>
    main()
  File "/opt/ros/humble/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 53, in main
    raise e
  File "/opt/ros/humble/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py", line 49, in main
    package = parse_package_string(
  File "/usr/lib/python3/dist-packages/catkin_pkg/package.py", line 786, in parse_package_string
    raise InvalidPackage('Error(s):%s' % (''.join(['\n- %s' % e for e in errors])), filename)
catkin_pkg.package.InvalidPackage: Error(s) in package '/home/sumedh/mav_platform_r2/src/diagnostics/diagnostic_common_diagnostics/package.xml':
Error(s):
- The manifest of package "diagnostic_common_diagnostics" (with format version 1) must not contain the following tags: exec_depend
- Either update to a newer format or replace <exec_depend> tags with <run_depend> tags.
CMake Error at /opt/ros/humble/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:95 (message):
  execute_process(/usr/bin/python3.8
  /opt/ros/humble/share/ament_cmake_core/cmake/core/package_xml_2_cmake.py
  /home/sumedh/mav_platform_r2/src/diagnostics/diagnostic_common_diagnostics/package.xml
  /home/sumedh/mav_platform_r2/build/diagnostic_common_diagnostics/ament_cmake_core/package.cmake)
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/humble/share/ament_cmake_core/cmake/core/ament_package_xml.cmake:49 (_ament_package_xml)
  /opt/ros/humble/share/ament_cmake_python/cmake/ament_python_install_package.cmake:60 (ament_package_xml)
  /opt/ros/humble/share/ament_cmake_python/cmake/ament_python_install_package.cmake:39 (_ament_cmake_python_install_package)
  CMakeLists.txt:10 (ament_python_install_package)

```

xkaraman, 2022-12-13:
@sumedhreddy90 You can checkout the `rolling` or even `galactic` branch and it should build fine or you can manually change the `buildtool` tag to `ament_cmake` and `ament_cmake_python` (some other changes to `export` tag are present and maybe need to be changed too for ROS2. check `rolling` branch of `package.xml`) and try again.

xkaraman, 2023-01-03:
Closing as this is resolved with #267.

PR #267 (2022-12-19): Fixing porting remainders
Fixes #262
closed 2023-01-03 by xkaraman