ros/diagnostics #273: Change default branch to Rolling

ralph-lange, 2023-01-25:
Hi @trainman419, since @ct2034 has created nice READMEs for diagnostics in the ROS 2 branches, we would like to ask whether it is possible to switch the default branch to rolling instead of noetic-devel.

trainman419, 2023-01-25:
Good idea! Done.

ralph-lange, 2023-02-06:
Hi @trainman419, sorry to bother you again on the default branch. @ct2034 and I decided to release diagnostics for Foxy, Humble, and Rolling together from a new branch named `ros2` (which we did a few days ago), but we also included the Git histories of the three separate branches foxy, humble, and rolling in this new branch as far as possible/reasonable.
Could you therefore please change the default branch to `ros2`? I'll rename `rolling` to `stale/rolling` then.

trainman419, 2023-02-06:
Done.

closed 2023-02-07 by ralph-lange