ros/diagnostics #293: Incorrect dependency export of pluginlib

ksuszka, 2023-03-06:
A few days ago our code which uses diagnostic_aggregator stopped compiling. After some digging it seems that the latest release of humble pacakges (20230221) includes [this change](https://github.com/ros/diagnostics/commit/c1988973f646614466b193eba1ab21f0b7f64ce9) which breaks packages with a dependency on the diagnostic_aggregator.

Here is a command to reproduce this error using docker (it creates an empty package with a dependency on the diagnostic_aggregator loaded through ament_cmake_auto helpers):
```shell
cat <<END_OF_DOCKERFILE | DOCKER_BUILDKIT=1 docker build --build-arg ROS_VERSION=humble -
# syntax=docker/dockerfile:1
ARG ROS_VERSION
FROM ros:\$ROS_VERSION
ARG ROS_VERSION
RUN apt-get update && apt-get install -y ros-\$ROS_VERSION-diagnostic-aggregator
RUN mkdir -p /ws/src/xxx

RUN <<"EOF" cat >> /ws/src/xxx/package.xml
<package format="3">
  <name>xxx</name>
  <version>0.0.1</version>
  <description>x</description>
  <maintainer email="x@x.xx"></maintainer>
  <license>BSD</license>

  <buildtool_depend>ament_cmake_auto</buildtool_depend>

  <depend>diagnostic_aggregator</depend>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>
EOF

RUN <<"EOF" cat >> /ws/src/xxx/CMakeLists.txt
cmake_minimum_required(VERSION 3.15)
project(xxx)

find_package(ament_cmake_auto REQUIRED)
ament_auto_find_build_dependencies()

ament_auto_add_library(${PROJECT_NAME}_lib SHARED test.cpp)

ament_auto_package()
EOF

RUN touch /ws/src/xxx/test.cpp

WORKDIR /ws
RUN . /opt/ros/\$ROS_VERSION/setup.sh && colcon build

END_OF_DOCKERFILE

```

Running this command fails with the error:
```
CMake Error at /opt/ros/humble/share/ament_cmake_auto/cmake/ament_auto_add_library.cmake:66 (add_library):
   Target "_lib" links to target "pluginlib::pluginlib" but the target was not
   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
   an ALIAS target is missing?
```

Changing ROS_VERSION to galactic in the first line makes it finish successfully.

ct2034, 2023-03-07:
Hi @ksuszka. Thanks for reporting the issue.
Since `diagnostic_aggregator` has no difference in the CMakeLists between humble and galactic, I assume the problem is `ament_auto_find_build_dependencies` not finding the `pluginlib` dependency 
I was able to fix your problem by manually adding a dependency on `pluginlib`: https://github.com/ct2034/ros-docker-playground/commit/ba90e15606f318c88c032030408ec365972f53d5
Could you please let me know if this approach works for you?

ksuszka, 2023-03-07:
Yes, adding pluginlib explicitly solves the problem.
However, your assumption that there is no difference between galactic and humble seems to be incorrect.
This code worked on humble with `ros-humble-diagnostic-aggregator/jammy,now 3.0.0-1jammy.20230118.000027`. The latest `ros-humble-diagnostic-aggregator/jammy 3.1.0-2jammy.20230221.145107` broke it. Galactic is still on `ros-galactic-diagnostic-aggregator/focal,now 2.1.3-1focal.20221208.065350` version, and it seems that [CMakeLists.txt for galactic branch is 3 years old](https://github.com/ros/diagnostics/blob/galactic/diagnostic_aggregator/CMakeLists.txt) and [CMakeLists.txt for humble](https://github.com/ros/diagnostics/blob/stale/humble/diagnostic_aggregator/CMakeLists.txt) was updated 4 months ago with this commit: https://github.com/ros/diagnostics/commit/c1988973f646614466b193eba1ab21f0b7f64ce9

ct2034, 2023-03-08:
You are absolutely right. :+1: 
I have looked into it, again.
Just to clarify: Can you also reproduce this error with `diagnostic_aggregator` from source? If so, can you please verify that #294 fixes it.

ksuszka, 2023-03-08:
Yes, I can confirm that #294 compiled on the latest ros:rolling and ros:humble images fixes the issue.

Thanks for addressing this!

commit 10307991 (2023-03-08): exporting dependency on pluginlib fixes #293 (#294)