ros/diagnostics #297: Why is the /diagnostics_toplevel_state ERROR when one of the diagnostics is STALE

Rayman, 2023-04-18:
[In the current code](https://github.com/ros/diagnostics/blob/12ba2ee06d2609286cd2f38a22f3a176dd051da0/diagnostic_aggregator/src/aggregator.cpp#L246) the toplevel state is only STALE when ALL the diagnostics are STALE. Example:

[image]

What I think is more logical is that the state is STALE when one of the diagnostics is STALE and none of them ERROR:
[image]

What do you think about this logic? I've implemented this in my fork, but it is a breaking change

ct2034, 2023-04-24:
Hi @Rayman. Thanks for your suggestion. Just to be clear: The stale state is set by the generic analyzer if no message was received within a given timeout: https://github.com/ros/diagnostics/blob/a80bd1c33786e5e5642f91b2b6016048f32fbf0e/diagnostic_aggregator/include/diagnostic_aggregator/generic_analyzer_base.hpp#L196
Which can be useful information on the actuality of an information.

If I get it correctly, your suggestion is to treat it in aggregation like the other levels and aggregate it in the group. There, I would honestly have a hard time to rate it in severity between the other levels. Currently, it is level 3 which reads as the highest priority. This means you would only see stale on the highest level, even if another item in that group is in the error state. This can not be the intended behavior. Changing these levels would be a __SERIOUS__ breaking change. 

What is your take on that?

asymingt, 2023-09-20:
What I find odd about the current behavior is that if you have three leaf diagnostics rolled up into a group, if two of them are marked `discard_stale` (bar and baz in the example below), if no status updates are received  on them, but `foo` is OK, I would intuitively think that the `part` group should also be OK. However, what I'm seeing is that `foo` currently errors out.

```yaml
diagnostics_aggregator:
  ros__parameters:
    pub_rate: 1.0
    path: 'robot'
    analyzers:
      part:
        type: 'diagnostic_aggregator/AnalyzerGroup'
        path: 'part'
        foo:
          type: 'diagnostic_aggregator/GenericAnalyzer'
          path: 'foo'
          find_and_remove_prefix: ['/foo:']
          num_items: 1
        bar:
          type: 'diagnostic_aggregator/GenericAnalyzer'
          path: 'bar'
          find_and_remove_prefix: ['/foo:']
          discard_stale: true
        baz:
          type: 'diagnostic_aggregator/GenericAnalyzer'
          path: 'baz'
          find_and_remove_prefix: ['/baz:']
          discard_stale: true
```

Rayman, 2023-09-21:
I did not want to propose to merge a breaking change in noetic, so I've implemented my proposed change in our fork: https://github.com/nobleo/diagnostics. Feel free to use it

I've implemented the change for the toplevel diagnostics and for the `AnalyserGroup`.

asymingt, 2023-09-21:
I also added my implementation here:
https://github.com/ros/diagnostics/pull/315

Timple, 2024-01-19:
Since this was a breaking change for Noetic, it probably also is for Humble and Iron?

Does it make sense to put some effort into this before ROS [Jazzy](https://dlu.github.io/ros_clock/index.html)?