ros/diagnostics #302: [Question] How to disable "individual" diagnostic messages

jesusramondovale, 2023-05-03:
Hi! I'm using this great package on my Ubuntu 22.04-ROS2 Humble in order to diagnostic my robot's hardware: 

    - Sick LiDAR scanner
    - Realsense Camera D455

jesusramondovale, 2023-05-03:
More info: I wanna see only the [0]th and [8]th item of the DiagnosticStatus sequence that is being published :

[image]

ct2034, 2023-06-12:
Hi @g-gemignani.

I think the best approach would be to write a custom aggregator: https://github.com/ros/diagnostics/tree/ros2/diagnostic_aggregator#analyzers

There is not really a mechanism to change the diagnostic topic itself. The assumption is, that the topic contains everything you may be interested in, and you can use the aggregator to make it more readable and increase interpretability.

Hope this helps. Otherwise, feel free to open another issue.

closed 2023-06-12 by ct2034

g-gemignani, 2023-06-12:
I guess the message was directed to @jesusramondovale  ;)
Thanks for the answer @ct2034!

ct2034, 2023-06-12:
Haha sorry. This was the wrong mention :)