ros/diagnostics #303: [ros2] Diagnostics aggregator base path

Noel215, 2023-05-04:
Hi,

With a diagnostic named `led1`:
```
analyzers:
  ros__parameters:
    path: Aggregator
    hardware:
      type: 'diagnostic_aggregator/GenericAnalyzer'
      path: Hardware
      startswith: [ 'led' ]
```

The name for that diagnostic  in the topic `/diagnostics_agg`  would be `/Aggregator/Hardware/led1` 

However, if I don't set the base path:
```
analyzers:
  ros__parameters:
    hardware:
      type: 'diagnostic_aggregator/GenericAnalyzer'
      path: Hardware
      startswith: [ 'led' ]
```
The name would be `//Hardware/led1`.
With the same case in ROS, the name would be `/Hardware/led1`

Is there any reason why the default `base_path` for the Diagnostics Aggregator in ROS 2 is `/`?

ct2034, 2023-05-24:
Hi @Noel215.

Thanks for the issue. 
I think https://github.com/ros/diagnostics/blob/369a769d2e8fd1babe97526f45d15628a76c0b0d/diagnostic_aggregator/src/aggregator.cpp#L67 is the culprit. 
Where in ROS1 the default value was https://github.com/ros/diagnostics/blob/12ba2ee06d2609286cd2f38a22f3a176dd051da0/diagnostic_aggregator/src/aggregator.cpp#L46.

I assume you would argue that the old default made more sense. I would agree.

Noel215, 2023-05-25:
Hi @ct2034,

Your assumption is correct.
I've just added a comment with a change to the PR because I cannot push to the repository.

PR #305 (2023-06-06): Empty default aggregator base_path
Fixes #303
closed 2023-06-06 by ct2034