ros/diagnostics #314: About the use of this diagnostic node package

TZECHIN6, 2023-09-12:
It seems this package focuses on hardware-level diagnostic, is it able to monitor the state of a node?

It is very common that a ROS project includes numbers of node, I am finding a way to check the status of node, and publish an error or warning when any of operational nodes is down.

I am very new to this package, and not sure will this overkill my purpose.

ct2034, 2023-09-27:
Yes, the `get_node_names()` method should return all running nodes. I am not sure, though, how quickly a node is removed after it dies. Maybe you have to test this.

TZECHIN6, 2023-09-27:
I can come up with a very quick way to test it using python list element check, but this come down to hardcoding the node name needed to be checked... which some time the launch file can rename the node. So I am not so sure is there any dynamic way to solve this if using `get_node_names()`.

I can share it to the common diagnostics, do you expect the output of the diagnostics result in a form of this `/diagnostic` topic? Acutally I havent use this before, so have to understand the structure of it..

ct2034, 2023-10-09:
You can publish the results similar to how it is done here https://github.com/ros/diagnostics/blob/ros2/diagnostic_updater/diagnostic_updater/example.py

closed 2023-10-10 by TZECHIN6