ros/diagnostics #386: diagnostic_updater binaries are not available for RHEL

christophfroehlich, 2024-07-17:
Since quite some time the build job for RHEL fails
https://build.ros2.org/job/Rsrc_el9__diagnostic_updater__rhel_9__source/
which causes our CI jobs to fail
https://github.com/ros-controls/ros2_controllers/actions/runs/9971295351/job/27552070956

```
08:33:29 DEBUG: Cloning into 'ros-rolling-diagnostic-updater'...
08:33:29 DEBUG: fatal: Remote branch rpm/ros-rolling-diagnostic-updater-4.3.0-1_9 not found in upstream origin
08:33:29 DEBUG: Child return code was: 128
08:33:29 ERROR: Command failed: 
08:33:29  # git clone --branch rpm/ros-rolling-diagnostic-updater-4.3.0-1_9 --depth 1 --no-single-branch https://github.com/ros2-gbp/diagnostics-release.git ros-rolling-diagnostic-updater
```

Is there a broken release? any solution for this? Thanks!

ct2034, 2024-07-17:
Hi @christophfroehlich. Thanks. I will look into this.

ct2034, 2024-07-31:
The issue is solved:
https://build.ros2.org/job/Rsrc_el9__diagnostic_updater__rhel_9__source/38/ ✅

closed 2024-07-31 by ct2034

christophfroehlich, 2024-08-11:
humble, jazzy, and rolling are fine now. But Iron still seems to have a problem?
https://build.ros2.org/job/Isrc_el9__diagnostic_updater__rhel_9__source/

christophfroehlich, 2024-08-22:
@ct2034 friendly ping :)