ros/diagnostics #387: Temporarily rolling back jazzy 4.2.0-1 update due to downstream issues.

marcoag, 2024-07-18:
@ct2034 I will temporarily roll back your latest rosdistro pr for jazzy: https://github.com/ros/rosdistro/pull/41859 due to a bunch downstream issues with the `diagnostic_updater` that are creating several [jazzy regressions](https://repo.ros2.org/status_page/ros_jazzy_default.html?q=REGRESSION). 

You can see failing "Downstream Projects" in: 

https://build.ros2.org/view/Jbin_uN64/job/Jbin_uN64__diagnostic_updater__ubuntu_noble_amd64__binary/

It looks like some symbols might be missing from the binaries so it would be great if you can take a look at this. Feel free to send an update PR anytime.

ct2034, 2024-07-18:
Hi @marcoag. Could you please help me identify one failed run of the downstream jobs. I looked through almost all of them and could not identify one that failed due to diagnostic_updater

marcoag, 2024-07-19:
Sorry, I didn't realized that after the rollback the downstream jobs would show up as successful and hard to find. Let me point some of them out:

https://build.ros2.org/view/Jbin_uN64/job/Jbin_uN64__swri_roscpp__ubuntu_noble_amd64__binary/24/consoleFull
https://build.ros2.org/view/Jbin_uN64/job/Jbin_uN64__swri_roscpp__ubuntu_noble_amd64__binary/24/consoleFull

The error show up like this:
```
/opt/ros/jazzy/include/diagnostic_updater/diagnostic_updater.hpp:378:(.text+0x6633): undefined reference to `diagnostic_updater::Updater::Updater(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeClockInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeLoggingInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeParametersInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTimersInterface>, std::shared_ptr<rclcpp::node_interfaces::NodeTopicsInterface>, double)'
```

Thanks for having a look btw.

ct2034, 2024-07-19:
I see. This is most likely caused by https://github.com/ros/diagnostics/pull/331. Without looking too deep into it. 

Unfortunately I am in the last hours before one week of vacations right now. If I don't find the time to do this over the weekend I can only look into it after July 30. Just FYI.

ct2034, 2024-07-19:
@Rayman could you please have a look at this if you have time. Most likely this just misses an include in the right place.

Rayman, 2024-07-22:
I've created a fix for the missing symbols: https://github.com/ros/diagnostics/pull/388

The `swri_roscpp` package actually has another bug. I'll send a fix for that.

To be honest, I'm a bit confused by the branches. It seems that `ros2` is behind `ros2-jazzy`.

ct2034, 2024-07-30:
@Rayman But, could you verify that this solved at least the above-mentioned linker problem?

ct2034, 2024-07-30:
> To be honest, I'm a bit confused by the branches. It seems that `ros2` is behind `ros2-jazzy`.

And I fixed that. There were some features not properly backported. But in the future, please make all PRs against `ros2`.

Rayman, 2024-07-30:
If you have any more build failure, feel free to ping me

ct2034, 2024-07-31:
see https://github.com/ros/rosdistro/pull/42285 for the release

closed 2024-07-31 by ct2034